plut0nium/0xB2 #3: About Bom

F1hone, 2022-07-26:
Hello，this is a great project

I want to try to make a board myself

Could you please provide the bom sheet related to this project?

This will help me a lot, thanks

bstiq, 2022-07-27:
Hi,

The latest gerber, BOM and POS files are in the release section of this repo.
Please note, at the moment the flash chip is not available at JLC (it's possible to order it but will take some time)

closed 2022-07-28 by bstiq

F1hone, 2022-08-08:
[image]
Yes！
 i made it 
nice design

F1hone, 2022-08-08:
[image]

bstiq, 2022-08-08:
Hi,

I'm glad you figured it out ! Enjoy the rp2040's power.

I'll close this issue, feel free to reopen it if there's any questions.

Here is the pinout if necessary:
[image]