plut0nium/0xB2 #4: Alternative NOR flash

toniz4, 2022-07-27:
Since the main model of NOR flash seems to be unavailable, there is other models with more or less storage, that are a drop in replacement?

bstiq, 2022-07-28:
At this point we don't know of any other flash that would be a drop in replacement.
You can either wait for restock, order the parts (that is possible at JLC), or  use a different part.

Part C97521 at JLC should be a suitable replacement, but is bigger and will require modifying the PCB.

Regards,

plut0nium, 2022-07-28:
In fact, there are several other chips that could be used, but none of them seem to be stocked by JLC at the moment.

Most QSPI NOR Flash would work if they are in the same form factor (USON-8) and support 3.3V.

plut0nium, 2022-10-07:
P25Q32H-UXH-IR (JLC part# C414068) has been used by some people.
Did not test it personally, though.

Ardakilic, 2023-01-13:
Hey @plut0nium , thanks for the information! Currently, there's P25Q32H-UXH-IR available on JLCPCB, and instead of XC6210B332MR-G, I ordered some PCBs with the alternative, which is 4 MBs, and will inform about how it'll go.

I already had connectors, and I'll hand-solder them, the rest should be same.

Ardakilic, 2023-02-14:
@plut0nium Heya, I don't know if this is related, but here comes my experiment:

Out of the box, even though the module uploads the stuff, unlike the sparkfun rp2040, the keyboard could not be identified.

I was frustrated, since I produced about a dozen (I knew the risk).

As a last resort, I wanted to try KMK and CircuitPython. I uploaded [the Sparkfun Pro Micro RP2040's circuitpython uf2 file](https://circuitpython.org/board/sparkfun_pro_micro_rp2040/) to both Sparkfun pro micro rp2040, and to splinky, and it identified for both! Also the NAND size was correct! This gave me hope.

After that, I re-uploaded the qmk uf2 file, and boom! it worked nicely!

I don't know how, or why, but installing CircuitPython uf2 beforehand helped. Maybe this was needed for factory-produced modules, I don't know. 

I'm currently testing the 5th module, all is same for every one of them

Ardakilic, 2023-02-15:
For the sake of the testing, I just found out this part is available on stock on JLCPCB:

MX25R8035FZUIL0 : https://jlcpcb.com/partdetail/mxic_macronix-MX25R8035FZUIL0/C152596

Which is a 8 megabyte module. I'll try it as well.

alinelena, 2023-06-14:
is this a replacement? MFR.Part # W25Q16JVUXIQ
JLCPCB Part # C2843335 to my untrained eye looks compatible... but any thoughts appreciated

original 
[image]
vs
[image]

Ardakilic, 2023-06-14:
Would these ones work also, it's bigger than 16 mbits, and are in stock as well: 

1. https://jlcpcb.com/partdetail/Zetta-ZD25WQ32CEIGR/C5258281
2. https://jlcpcb.com/partdetail/puya-P25Q32L_UXHIR/C559225

bstiq, 2023-06-15:
The first one you linked seems to have the same pinout, I couldn't check the nominal voltage on a quick check.
So it seems like it could work, you might need to modify the QMK code to accommodate for a different type of memory.
Same thing for the others - I haven't tested them though.

If you are buying in small quantities, I recommend also looking at off-the-shelf solutions like the liatris or even kb2040 if you don't need the bottom pins

alinelena, 2023-06-15:
C2843335, W25Q16JVUXIQ is 3V
C5258281,ZD25WQ32CEIGR, is 1.65 - 3.6V
C559225,P25Q32L-UXH-IR, is 1.65 - 2V
C414068, P25Q32H-UXH-IR is 2.3 - 3.6V this is reported as working.

Ardakilic, 2023-06-15:
Since @plut0nium mentioned if they work with 3.3v, and the pinouts match, I assume 

[C5258281,ZD25WQ32CEIGR, is 1.65 - 3.6V](https://jlcpcb.com/partdetail/Zetta-ZD25WQ32CEIGR/C5258281)

Should work.

If I create a new batch for my prototype, I'll try this one.

alinelena, 2023-06-17:
> C2843335, W25Q16JVUXIQ is 3V

just to note waveshare is using this in one of theirs
https://www.waveshare.com/rp2040-tiny.htm

alinelena, 2023-06-20:
looking more into this one would expect instruction sets to match too 
https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/boot_stage2/boot2_w25q080.S but this is beyond me

plut0nium, 2023-09-18:
Thanks for the feedback!

I have been considering making a list of NOR chips reported working with this board... but I do not produce or sell pcbs, so have a very limited view of what's really happening right now.

wlhlm, 2023-09-30:
I had some Splinkys made with P25Q32H-UXH-IR (LCSC C414068) flash. To make QMK work I have to set
```c
#define RP2040_FLASH_GENERIC_03H
```
in `config.h`.