plut0nium/0xB2 #5: About KMK

F1hone, 2022-08-10:


plut0nium, 2022-08-10:
Sorry, I don't know anything about CircuitPython and/or KMK...

But have you tried with the Sparkfun RP2040 version of CircuitPython: https://circuitpython.org/board/sparkfun_pro_micro_rp2040/ ?

F1hone, 2022-08-10:
It doesn't work,
I tried other rp2040 boards and none of them worked
Maybe I need to compile a uf2 file myself

plut0nium, 2022-10-07:
> I tried other rp2040 boards and none of them worked

If I understand correctly, this issue is not specific to this board.

mitswan, 2022-11-07:
Has anyone found a workaround for getting Circuitpython on this? Having the same challenge

mitswan, 2022-11-08:
For what its worth - the main difference is the flash chip GD25Q80C is not yet supported by circuitpython

bharathyes, 2024-01-26:
@mitswan where best to check / track when support for this board is added ?