plut0nium/0xB2 #8: Question about the connector

Ardakilic, 2023-01-14:
Hello,

First of all, thanks for making this project available for us all! 

I'll solder the connectors myself, and ordered the rest from JLCPCB, however I have a question. 

At the current state, the connectors are mid mount, so I was wondering would either of this as a connector work? The connectors I have are through hole, so I wanted to buy a batch, but wanted to ensure they fit properly.

https://www.aliexpress.com/item/32998900371.html

[image]

Thanks in advance!

bstiq, 2023-01-18:
Hey, 

From a quick check on the link you posted, the dimensions look close. There's a few places where they slightly differ, so you should also check both datasheets and maybe put one footprint over the other to check (in Kicad).

It looks like you might need to change the position of the pads, and the shape of the edge cut.

The datasheet on the link is too pixelated to check the pads, so you should find a pdf and check the numbering carefully (A1...A12). They have to match, so you might need to re-route some connections.

That's as far as I can see, others can feel free to chime in.

Ardakilic, 2023-01-26:
Hello @bstiq , thanks for the feedback.

Currently the PCBs are being produced, and since it's Chinese new year, it's delayed, so I cannot exactly tell how it will be. Just in case I ordered from both variants and will check the compatibility, risking sacrificing either one. 

I will update here after I try it, I'd also appreciate if anyone has experience with these.

Thanks again!

Ardakilic, 2023-02-14:
Heya, I just got the boards today, I ordered without connectors to make it cost cheaper:

[image]

I had bought these and they were delivered already, so I just soldered them.

The 1.6 mm mid mount ones work awesome!

This is my hand soldering:

[image]

[image]

The other one would have also worked electronically and mechanically, but the 1.6mm was looking more proper for this.

Hand-soldering these, and using stamp holes instead of v-cuts made it cost about $20 cheaper overall (it could be assembled as cheaper level like this because of lack of a through hole component and lack of v-cut). 

I hope a panel like my modified one (which JLCPCB did actually) is considered from the repo authors.

closed 2023-02-14 by Ardakilic

bstiq, 2023-02-15:
Hey ! Thanks for all the feedback.

Regarding the assembly costs: on my side, I use a usb-c port that requires standard assembly. I also have stocks of components and to use those I also need standard assembly. In that sense, I'm not concerned by a mousebites panel.

That's something that's also relatively easy to do in Kikit, so unless @plut0nium wants to do that specifically I think it's going to stay an option.

If you want to create a PR to add eg. the kikit options that's possible of course.

Ardakilic, 2023-02-15:
Sadly I don't know how to modify a gerber / project with KiCad and KiKit (which I believe is [this ](https://github.com/yaqwsx/KiKit) I guess) 😞

I could always upload the JLC's updated gerber to here, but I guess it will be futile since panel is generated through scripts and if you ask nicely I guess JLC could do it anyways.

I'll try to check and alter the panel if it's done through a script, I hope I can put it together with ease, but don't keep your hopes up.