plut0nium/0xB2 #9: Suggestion: Make a Stamp Hole Panel Variant

Ardakilic, 2023-01-28:
Hello,

I'm currently producing this board, and I choose economics for parts placements, considering I will solder the connectors myself. When I do like it, the v-cut makes it 20 usd more expensive for production.

As an alternative, JLCPCB suggested using stamp holes, and updated my gerber like this:

[image]

So, when I decide to solder the connector myself, and use stamp holes instead of v-cut, it's almost $1 cheaper per module, just because of the selections.

I'd appreciate if this could be considered for future releases.

Thanks again for making this board available, can't wait to hold on the prototypes 😊

Ardakilic, 2023-02-14:
This all worked nicely. I got all of the assembled boards without issues, and all are running nicely. I hope this is considered for the future releases!

alinelena, 2023-06-14:
the other option is to let jlcpcb panelize them for you... bad news is you get 12 on a panel...

plut0nium, 2023-09-18:
Thanks for the feedback.

However, there are many panelizing options, some are more suited for specific uses (auto vs manual PCB assembly, cost optimization...).
The panelized version in the release is just provided for convenience, but the main deliverable remains the single-pcb cad files.

People are encouraged to panelize themselves to meet their application requirements.

closed 2023-09-18 by plut0nium