plut0nium/0xB2 #12: are gerbers ok (v3.2)

alinelena, 2023-06-17:
I am looking into printing few of these i upload the single board and choose panelise by pcb 4x2 but then when they show me the parts placements looks like this, notice usb type c not in place... similar if I choose the panelized version from the repo.

[image]

also i notice my replacement nor does not show.. i went with this, C5258281, this check with them

plut0nium, 2023-06-19:
The bad placement of the USB-C is a common issue, and is usually solved by JLCPCB operators during the final review.

Many people have used the gerbers without reported issue.

alinelena, 2023-06-19:
thank you! I assumed show but I wanted to have it triple checked.

closed 2023-07-12 by alinelena