plut0nium/0xB2 #15: Which version of CircuitPython to be flashed ?

bharathyes, 2024-01-17:
Currently building a Lily58 PRO with Splinky as the controller. 

Stuck at writing the firmware to the board.

Which version of CircuitPython needs to be flashed for setting up KMK ?

From the [firmware downloads page](https://circuitpython.org/downloads?q=2040) I have tried the following : Adafruit KB2040, Sparkfun Pro Micro RP2040 and even the Raspberry Pi Pico.

None of the seem to work and the device boots back named just RPI-RP2 rather than CIRCUIT-PY or similar as it does in the RPI Pico board.

Any help on setting up the Lily58 keyboard would be appreciated.

plut0nium, 2024-01-18:
According to issue #5 this depends on the model and characteristics of the NOR flash chip that was used.

I do not have any experience with CircuitPython and/or KMK, so I'm afraid I can't provide any support on this matter.
This repository is dedicated on the hardware design of the controller board.

bharathyes, 2024-01-26:
Okay thanks. Will follow that issue to keep track of when support is added.

closed 2024-01-26 by bharathyes